skybrush-io/live #1: Show maximum altitude reached by drones after loading the show file

ntamas, 2022-05-02:
A common question that we get from people using the software is how they can check the maximum altitude that a drone will reach during a show. The usual response so far is to load the show in Skybrush Viewer and look at the altitude profile in the validation screen, but we should probably show the maximum altitude in the Show control panel as well.

My idea would be to put it in the secondary label of the button that allows the user to load the show file so it would become something like:

"123 drones, 4:56, max AGL 78m"

Isti115, 2022-05-05:
Idea implemented in #3.

ntamas, 2022-05-06:
Fixed in https://github.com/skybrush-io/live/commit/58e17fe4750b4ed5e3f4559e93747cf713e62ba7

closed 2022-05-06 by ntamas