skybrush-io/live #23: Add a control window when the Turn Off button is pressed within SkyBrush Live , to prevent unexpected situations

DavideLentini, 2022-10-08:
With the recent developments of a system for a remote on/off , I believe it is critical to add a control window within Sky Brush Live when pressing the Turn Off button .

A control window that allows you to cancel sending the Turn Off command in broadcast , or to confirm it . 

It might also be useful to add this control window also when pressing the Sleep or Wake Up command , but we have confirmed that the latter messages(wake up and sleep )are ignored if the drone is armed , so I do not think it is fully necessary . 

Thanks all !

DavideLentini, 2023-09-26:
Hi !
It would be great to have this option in the next version of Live . I find it very important for fleet safety . Thank you

ntamas, 2023-10-30:
Sorry, I haven't forgotten about this but I couldn't squeeze this into 2.8.0. I'll try to get this done before the next minor release.

ntamas, 2024-01-02:
Going through the issue tracker to clean things up. This is implemented in version 2.8.0 -- see attached screenshot.

[image]

closed 2024-01-02 by ntamas

DavideLentini, 2024-01-02:
Thanks @ntamas I appreciate it very much . happy new year