skybrush-io/live #48: Skybrush Live Going blank after starting

bytesByHarsh, 2024-11-12:
Just after starting the application, it loads for a few seconds but then the screen turns white. 

Version: 2.8.1
System: Ubuntu 20

[image]

isti115, 2024-11-12:
Hi, @bytesByHarsh!
I'm fairly certain that this is caused by an Electron / GPU driver version conflict, which results in the crash of the renderer process. Could you please complete the following steps in order to help diagnose and resolve issue?
- Check if your drivers are all up to date
- Share the make / model of your GPU
- Run the application with / without hardware acceleration enabled
- Run the application from a terminal window and share the output

bytesByHarsh, 2024-11-13:
Hi @isti115 
I have NVIDIA GeForce RTX 3070 in my system, and all the drivers are up to date.

After running in the terminal I am only getting the following line:

```bash
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0
```

I am not sure how to test with / without hardware acceleration enabled, if there is any specific command to do so I'll try it out.

bytesByHarsh, 2024-11-19:
Hey, thanks for the build,

Yes now it seems to be working fine, this binary is working even without specifying `ELECTRON_OZONE_PLATFORM_HINT=wayland`

And yes I am running the default Ubuntu desktop environment, with Intel Core i7 and Nvidia GeForce RTX graphic card.

ntamas, 2024-11-19:
Okay, so basically it seems like we need to upgrade Electron and then we should be okay. I've been planning a new Live release for quite a while now so that's one more reason to do it.