skybrush-io/live #73: Range selection with numeric hotkeys (e.g. s100-200)

vasarhelyi, 2025-06-24:
In the current version it is possible to enter a drone ID (e.g. 100) or a show ID (e.g. s100) + Enter to select a single drone.

Would be nice to extend this "regex" with handling multiple drones properly for selection. Possible use cases:

"s100-200" + Enter would select all drones with their show IDs ranging between 100 and 200 (all edges included)
"s100,150,s200" or "s100;150;s200" or perhaps even "s100 150 s200" + Enter would select drones with show IDs 100 and 200 and the one with physical ID 150
"s10-20,30-40" + Enter would select drones between show IDs 10-20 and drones with physical IDs between 30-40 (can be overlap between the two ranges so needs to be handled as a set not as a list)
etc.

ntamas, 2025-06-24:
Let's try to keep the scope minimal - no one asked for all these patterns and I really hardly see any use-case for the more complex ones, but it makes the implementation more complex. (Also, adding new features is easy, but _maintaining_ them in the long term is hard and time-consuming).

I'd restrict ourselves to implementing `s100-200`, i.e. a dash followed by a second number. Maybe if we really want to add support for selecting multiple drones, a better approach would be to add support for prefixing a selection with `+`, which would then _add_ the newly selected drones to the existing selection instead of replacing the selection.

volfpeter, 2025-06-26:
The `+` prefix feels useful. But if we add it to mutate the selected indices, then for sake of completeness, I would also add the `-` prefix to remove items from the selection. If we add that as well, then using `-` for the "range" separator as well could be a bit confusing, so maybe we could use `:` for that (it's very common). This way, supported expressions would look like this: `s31:42`, `+s22`, `-s23-45`, `10-20`, `-10`, `+65:70`.

I'll need to check the code to see how we're handling keyboard events for this feature. It may be best to only add range selection now, and support for `+` and `-` later.

ntamas, 2025-06-30:
#76 is now merged so closing this.